chriskinal/AiO_New_Dawn #16: PWM motor drive seems to be acting like Cytron

chriskinal, 2025-07-15:
The LEDs while the sim is driving straight with AS engaged, swing your WAS back and forth. The PWM1/2 LEDs don't light symmetrically, and the MA/MB LEDs don't either. Then when disabling AS, something MA stays on.

with the sim driving centered on a straight line, the further your WAS is off center the brighter the PWM1/2 and MA/MB respectively should be. (with minPWM 0 and P gain fairly low) Right when your WAS is centered they should all be off but there should be very little deadzone before one or the other PWM/Mx LEDs start dimly lighting again

all PWM and Mx LEDs should be off when AS is off

PWM1 output appears digital, not analog

I think it's controlling the motor driver like a cytron (PWM & DIR), but we're using a diff version of the chip. We need PWM1 output for one direction and PWM2 for the other direction.

Current implementation:
/Users/chris/Documents/Code/Firmware_Teensy_AiO-NG-v6/include/AutosteerPID.h motorDrive() Line 66.

MotorDrive is called right after calculatePID().

m-elias, 2025-07-15:
Because this board has the on-board DRV8701P, the Cytron/IBT2 selector in AOG should be ignored. Rather, check for Keya, then Danfoss, then default to the on-board DRV8701P.

a. Keya Canbus
Presence of 0x07000001 messages on Canbus 3
b. Danfoss
PGN251 Byte 8 = 0x01
c. On-board DRV8701P

a. Keya Canbus - Send Keya commands as currently in the Keya motor driver.
b. Danfoss - Two machine outputs (Output 5 & 6). Output 5: HIGH to enable steering. Output 6: 50% = Center, 100% = Full Left, 0% = Full right
c. DRV8701P - PWM1 = Direction A & PWM2 = Direction B, in-active PWMx should be LOW
The Left/Right direction isn't critical at this point, direction can be reversed in AOG UI but once someone tests with an existing AOG config they can inform whether those need swapping to maintain compatibility.

chriskinal, 2025-07-15:
OK. The notes in the old firmware give the following Danfoss percentages;
50% = Center, 25% = Full Left, 75% = Full right.
https://github.com/AgOpenGPS-Official/Firmware_Teensy_AiO-NG-v6/blob/fa89e5519faa35de0c420e30b8913d9b04a1cb02/include/AutosteerPID.h#L51

chriskinal, 2025-07-16:
Motor drivers using new architecture. Keya, Danfoss and DRV8701 are supported.