chriskinal/AiO_New_Dawn #29: Danfoss output skips a pulse while AS is active

m-elias, 2025-07-22:
I'm seeing some skipped PWM pulses on the Us (section 6?) PWM output while autosteer is active. It looks like every 14-15 pulses it outputs a zero duty-cycle pulse.

[image]

When AS is inactive/disengaged the 50% duty-cycle output is nice and steady.

[image]

chriskinal, 2025-07-23:
Fixed by optimizing the Danfoss PWM updates to reduce I2C traffic and eliminate pulse skipping

chriskinal, 2025-07-23:
Fixed in v0.1.43.

closed 2025-07-23 by chriskinal