chriskinal/AiO_New_Dawn #39: Expand Work input to read analog height sensors

m-elias, 2025-07-29:
The Work input is an analog input with an op-amp voltage follower buffer. By default it has a pullup resistor in circuit so that it mimics the older steer modules' Work inputs (connect to gnd to activate the input/function). By connecting an analog voltage reference such as a combine feed house height or cultivator/planter height sensor (and disconnecting the on-board pullup as required) it can trigger the Work signal sent to AOG on & off. 

Here's how I set it up previously in the Mongoose UI

[image]

[image]

[image]

The main bit is to have a web UI customizable threshold and hysteresis which sets the analog level to trigger the on & off state of the Work input. In the case of these screenshots, I used the Work Switch/Analog Input title bar, the ON/OFF indicator and the Input progress bar's background colors to visually indicate the Work input's state of ON or OFF. The progress bar has the additional grey color to show when the analog input is within the hysteresis range and the Work state is therefore unchanged. The Set Analog Threshold button saved the current analog level as the input threshold whereas the Set for Digital Input button set a 50% threshold and a wide hysteresis (similar to a Schmidt trigger). The Dec & Inc buttons where there to fine tune the threshold and a hyst dropdown to select a from a few ranges of hyst options. The Invert toggle is there to invert the final state logic.

I think there is a potential use case to employ an additional Invert option on the input so that the analog gauge & progress bar could be inverted as needed to suit the operator's preference but it wouldn't be necessary for successful operation. 

I would like to see a single analog gauge/bar/graph to show the analog input with markers indicating the hysteresis range which mathematically would be threshold - (1/2 hysteresis) and threshold + (1/2 hysteresis). Along with changing the background color to green, red & grey accordingly I think it should be quite self explanatory. 

[image]

chriskinal, 2025-08-01:
What do red, green and gray represent? 
Is this threshold value sent / received in a PGN to / from AOG?

chriskinal, 2025-08-02:
I see there is a set button. So the sensor on the combine will supply the voltage for the reading on the bar. Need a button to "set" or capture that ADC reading as the set point. Now for hysteresis, would just be an input box or dropdown list where you type in or select the total hysteresis around the set point?

m-elias, 2025-08-02:
A slider would also work for hyst but it's probably best to limit the range of values to a reasonable range, maybe +-5% to 25%. Maybe the "Set for Digital" button should be cashed "reset to default" instead

chriskinal, 2025-08-02:
And what value is default? 50%?

m-elias, 2025-08-02:
The default threshold should be 50% with a large hyst (about 20%) so that it's like a Schmidt trigger input for the traditional digital mode.

chriskinal, 2025-08-02:
Does this "By connecting an analog voltage reference" use the same WORK_PIN currently used for a switch?

m-elias, 2025-08-02:
Yes, see my first message for all the details

chriskinal, 2025-08-02:
Are there other changes to the board required besides cutting the pull up SJ? I am seeing a pretty limited voltage / ADC range when using the WAS 5v. Min ADC raw of 64 and max 1686.

m-elias, 2025-08-02:
It's setup for 12v input

chriskinal, 2025-08-02:
Yeah, I just figured that out when I looked at the schematic really hard. ;-)

m-elias, 2025-08-02:
I could probably do the code except I'm not up to speed on adding it to the UI.

chriskinal, 2025-08-03:
Checkout v0.1.66 in the feat/analog_ws branch. Not as fancy as yours but functional.